vortexntnu/vortex-asv #21: asv simulator failed to run

ronjakr, 2022-01-26:
[image]

Most likely due to not having gazebo on computer

chrstrom, 2022-01-29:
Appears to be an error with allocation for displaying graphics. Need to look into the source of "x error of failed request badalloc (insufficient resources for operation)"

ronjakr, 2022-01-29:
[image]
Simulator runs on a clean ros melodic image, with the required components installed. Issue may lie in the compose file for the simulator

chrstrom, 2022-01-29:
Update: this may actually be an AMD mobile graphics problem

chrstrom, 2022-03-09:
Containerized simulator runs headless so this won't be a problem 🚀

closed 2022-03-09 by chrstrom