vortexntnu/vortex-asv #114: [TASK] Landmark Server V2

EirikKolas, 2023-11-19:
### Description of task

Create a landmark server, a a common interface for SITAW nodes to communicate with AS nodes

### Suggested Implementation
Create a ROS2 message on the form:

Message Type              | Name           | Description
------------------------- | -------------- | ----------- 
**Metadata**              |                | _Stuff to define what the object is_
`std_msgs/Header`           | header         | For reference frame and timestamp of the landmark
`string`                    | namespace      | For specifying the type of the landmark. (boat, buoy, etc)
`int32`                     | id             | Specific item within namespace. (namespace + id should uniquely define the landmark)
`int32`                     | action         | add/remove
**State**                 |                | _Stuff to define where the object is_
`int32`                     | type           | enum for type of object (dynamic/moving or static)
`geometry_msgs/Pose`        | pose           | Position and orientation of the object
**Extent**                |                | _Stuff to define how big something is_
`shape_msgs/SolidPrimitive` | shape          | Shape of the object. (Prism/Sphere)
 
All message types can be found [here](https://github.com/ros2/common_interfaces).

The old implementation of the landmark server requires the user to change the code for each new type of landmark added. The new implementation should be agnostic to type of landmark. The idea is for the AS side to use ROS actions to filter what kind of landmarks they want to listen to. E.g. filter for static/moving objects, only boats, only buoys etc.   

The landmark server should also support publishing a map with the `nav_msgs/OccupancyGrid` message. 

### Suggested Workflow

- [ ] Look at [old landmark server](https://github.com/vortexntnu/vortex-auv/tree/development/mission/landmarks) (or not)
- [ ] Create the message in the [Vortex-msgs repo](https://github.com/vortexntnu/vortex-msgs)
- [ ] Find a suitable way to store the landmarks, with ways to update, add and delete existing landmarks.
- [ ] Create a simple publisher to publish everything
- [ ] Create an [action](https://docs.ros.org/en/humble/Tutorials/Beginner-CLI-Tools/Understanding-ROS2-Actions/Understanding-ROS2-Actions.html) to filter what messages you want to listen to. The feedback topic should contain the filtered messages in some form (@vortexntnu/autonomous-systems decides what format this format should be in)

### Specifications

- [ ] Server can store any type of landmark without having to recompile/edit code.
- [ ] ROS Action to publish a filtered subset of the landmarks.
- [ ] ROS Action to get an occupancy grid representation.

### Contacts

- @EirikKolas, has some opinions
- @alekskl01

### Code Quality

- [ ] Every function in header files are documented (inputs/returns/exceptions)
- [ ] The project has automated tests that cover MOST of the functions and branches in functions (pytest/gtest)
- [ ] The code is documented on the wiki (provide link)

jorgenfj, 2024-02-27:
I closed to to await progression on the target tracker to better understand what the actual requirements for the server was. Should be good now though and the pull request can be reopened

EirikKolas, 2024-02-27:
noice!

closed 2024-03-03 by EirikKolas