vortexntnu/vortex-asv #164: [TASK] Implement real inertia and damping matrix for Freya

Andeshog, 2024-03-28:
### Description of task

The system matrices M and D for Freya should be improved to get values that better reflects reality.

### Suggested Workflow

-[ ] Take a look at the Otter of Maritime Robotics, as it is similar to Freya. https://github.com/cybergalactic/PythonVehicleSimulator/blob/master/src/python_vehicle_simulator/vehicles/otter.py

### Code Quality

- [ ] Every function in header files are documented (inputs/returns/exceptions)
- [ ] The project has automated tests that cover MOST of the functions and branches in functions (pytest/gtest)
- [ ] The code is documented on the wiki (provide link)

alekskl01, 2024-04-07:
Is this finished?

Andeshog, 2024-04-07:
> Is this finished?

Not yet, will try to finish it today

closed 2024-04-08 by Andeshog